drogue-iot/drogue-device #1: Use of `&'static mut self` is unsound

Dirbaio, 2021-02-04:
Very interesting project! :)

One thing that jumped at me while reading the blogpost is the use of `&'static mut self`. It's not sound since it allows the user to create multiple concurrent &mut's to the same data. For example you can "hold on" to self by storing it in a static. In the next call you get two &mut's to the same data, one from the static and another from self.

```rust
use cortex_m::interrupt::{Mutex, free};

static SAVED: Mutex<Cell<Option<&'static mut Foo>>> = Mutex::new(Cell::new(None));

impl NotifyHandler for Foo {
    fn on_notify(&'static mut self, message: State) -> Completion {
       free(|cs| {
           if let Some(self2) = SAVED.borrow(cs).get() {
                info!("Got two &mut's to the same data! {:?} {:?}", self, self2);
           } 
           SAVED.borrow(cs).set(Some(self)
       })
    }
}

bobmcwhirter, 2021-02-04:
Yep, discussion on reddit pointed us at that. We're adjusting it so that `self` moves around, hopefully making it more sound. Should be pushing shortly.

bobmcwhirter, 2021-02-04:
Also, we borrowed your `signal.rs` but I think @lulf got in touch about the copyright on that, yeah?

Dirbaio, 2021-02-04:
> We're adjusting it so that self moves around, hopefully making it more sound. Should be pushing shortly.

Nice :) 

> Also, we borrowed your signal.rs but I think @lulf got in touch about the copyright on that, yeah?

Yeah, it's no problem, that's what opensource is for. You didn't even need to ask but it's appreciated.

bobmcwhirter, 2021-02-04:
HEAD has been updated, everything has changed (at least a little), hopefully more soundness, but absolutely makes the blogpost out-of-date wrt specifics.

lulf, 2021-05-19:
Unsoundness should be gone now!

closed 2021-05-19 by lulf