drogue-iot/drogue-device #9: Create atomics crate to avoid hard dependency on atomics

lulf, 2021-02-15:
This is needed for drogue-device to work on some systems like cortex-m0. bbqueue can be used as an example to pull code from.

Dirbaio, 2021-03-23:
This should do the job :) https://github.com/embassy-rs/atomic-polyfill

lulf, 2021-03-24:
Excellent. Since this issue was created, I imported the wrappers from bbqueue to gain cortex m0 support, but I think we should depend on the atomic-polyfill crate instead.

dak-x, 2021-03-25:
The polyfill crate lacks implementation for compiler_fence. What the compiler fence does is just prevent a thread from racing with its signal handler. Since what the crate currently does is just block all interrupts for the crit-section on thumbv6.. target. Shall I add a void 'compiler-fence' inside the polyfill. Ill leave it with the necessary warnings. @Dirbaio?. I could also remove its usage inside the said targets to fix. #19 . I hope this is a viable solution and Im not missing anything here.

lulf, 2021-03-25:
@dak-x I don't think you need to replace compiler_fence or any references to core::sync::atomic::* types that already works on thumbv6 (I think compiler_fence is already supported on thumbv6).

Dirbaio, 2021-03-25:
The goal is to be a simple drop-in replacement, so please file issues for missing things! :) https://github.com/embassy-rs/atomic-polyfill/issues/1