drogue-iot/drogue-device #58: Want to compile drogue-device\examples\stm32l0xx\lora-discovery - always get this error...

VolITSol, 2021-08-09:
Hi,

I want to test the 

   Compiling drogue-device v0.1.0 (C:\Users\cvoller\git\drogue-device\device)
   Compiling serde_yaml v0.8.17
   Compiling stm32-metapac-gen v0.1.0 (https://github.com/embassy-rs/embassy.git?rev=c458ad52e698ae0e9810e90d094394bd3ab35c77#c458ad52)
   Compiling stm32-metapac v0.1.0 (https://github.com/embassy-rs/embassy.git?rev=c458ad52e698ae0e9810e90d094394bd3ab35c77#c458ad52)
   Compiling embassy-stm32 v0.1.0 (https://github.com/embassy-rs/embassy.git?rev=c458ad52e698ae0e9810e90d094394bd3ab35c77#c458ad52)
error: failed to run custom build command for `embassy-stm32 v0.1.0 (https://github.com/embassy-rs/embassy.git?rev=c458ad52e698ae0e9810e90d094394bd3ab35c77#c458ad52)`

Caused by:
  process didn't exit successfully: `C:\Users\cvoller\git\drogue-device\examples\stm32l0xx\lora-discovery\target\release\build\embassy-stm32-cce071716c42b90a\build-script-build` (exit code: 101)
  --- stderr
  thread 'main' panicked at 'failed to execute gen.py: Os { code: 2, kind: NotFound, message: "Das System kann die angegebene Datei nicht finden." }', C:\Users\cvoller\.cargo\git\checkouts\embassy-9312dcb0ed774b29\c458ad5\embassy-stm32\build.rs:20:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

lulf, 2021-08-09:
@VolITSol Embassy (on stm32) uses python to generate some of the peripheral at build time. I'm wondering if the NotFound comes from it not finding python3 executable in the system %PATH%?

lulf, 2021-08-23:
@VolITSol Hi, the python dependency should now be removed in embassy, so you might want to try pull the latest changes from main and try again. Please reopen if the issue still persists.

closed 2021-08-23 by lulf