drogue-iot/drogue-device #238: Improve getting started experience

lulf, 2022-03-16:
Current issues where we can improve

* Taking the step from example to standalone crate (templates)
* Simpler more understandable examples for basic things like blinky
* Better example documentation both in code and readme (now included inline in book as of #237)

@ctron Iirc, you had some more points around actor documentation. Could you add those points here? There are some actor  introcution in the 'Introduction', and a programmatic introduction in the 'Basic application' section of https://book.drogue.io/drogue-device/dev/index.html. But iirc you had some other valid points.

ctron, 2022-03-17:
What I guess would help is:
* Add some documentation on the actor model of Drogue Device
  *  What is an actor
  * What is the lifecycle? Init, Inbox, ...
  * When to use an actor vs when not to
* A basic tutorial, which builds (step by step) a simple device (like a blinking LED with a push-button that changes its behavior). Maybe pointing to a dedicated GitHub repository, which contains the full code (and only that). Ready to check out, build, and flash on a device we choose (like the microbit).

ctron, 2022-03-17:
Ok, after re-checking the docs. I think it already is there.

lulf, 2022-03-21:
Improvements:

* Added `cargo xtask clone` in #244 
* Added docs for BSP in #245 

Will take a stab at improving the code docs.

lulf, 2022-03-23:
I'm going to attempt to improve the experience of writing and running actors as follows:

1. Introduce a `spawn_actor!` macro to simplify creation and mounting of the actor context + actor:
```
let address = spawn_actor!(spawner, BUTTON_A, Button::new(board.btn_a));
```

2. Introduce a `actor` proc macro to simplify the Actor impl:

```
#[actor]
impl Actor for MyActor {
    type Message<'m> = MyMessage;
    async fn on_mount<'m, M>(&'m mut self, address: Address<Self>, inbox: &mut M)
        where M: Inbox<Self> + 'm {
        loop {
            if let Some(m) = inbox.next().await {
                // ... handle message
            }
        }
    }
}
```

The purpose of this macro is not to remove all the boilerplate, but rather prepare the actor implementations to require minimal modifications when async traits are fully available in Rust, so that just removing the #[actor] macro (and making the macro a dummy) will allow the code to keep working.