drogue-iot/drogue-device #330: Add MQTT client

overheat, 2022-12-17:
It's a quick common IoT protocol. Any road-map to support it?

lulf, 2022-12-18:
You can use https://crates.io/crates/rust-mqtt. I think we're just missing the example, which got removed a while back when the nrf + esp8266 example got removed. It would be nice to have this added back for one of the "IP-native" examples.

overheat, 2022-12-18:
> You can use https://crates.io/crates/rust-mqtt. I think we're just missing the example, which got removed a while back when the nrf + esp8266 example got removed. It would be nice to have this added back for one of the "IP-native" examples.

I found that hash `d9f15cf97cc7de75f815567cd9c625d301328c6e`. But looks a lots different than now, like the `networking` folder under `device/` is missing now. It's not easy for me.

Do you or any member available to add it back?