drogue-iot/drogue-device #331: examples/std/cloud not works.

overheat, 2023-01-05:
First the IP address look changed to `[Ipv4(Address([34, 249, 85, 182]))` and `Ipv4(Address([34, 247, 113, 130]))]`

But it not works if change to right IP address.
```
...
INFO  cloud] Response status: Forbidden
```

lulf, 2023-01-05:
Hmm, I think we need a proper DNS client now that the endpoints moved to AWS load balancers, which may change IPs.  I'll see if we can get a fixed IP in front of it that might help.

overheat, 2023-01-05:
@lulf BTW, do you have time to test this example? It not works on my testing, even change to right IP address.

lulf, 2023-01-05:
I updated the IP and pushed a change to the main branch, and the GH action that runs this example is passing, so it should work:

https://github.com/drogue-iot/drogue-device/actions/runs/3846267558/jobs/6551369367

overheat, 2023-01-15:
> Hmm, I think we need a proper DNS client now that the endpoints moved to AWS load balancers, which may change IPs. I'll see if we can get a fixed IP in front of it that might help.

Looks IP address changed again. A proper DNS client is needed, I think.