drogue-iot/drogue-device #353: Unable to discern whether to use boards or device/src/bsp/boards

abrudner, 2023-03-12:
Hi there,

I'm looking to use this toolset and I'm having trouble figuring out which BSP to use. It seems like the top level boards file is newer but only by a month, and there's only a few minor differences I can tell (looking at the STM32L4 directory). Would you please be able to provide some guidance?

lulf, 2023-03-13:
Sorry about that, the bsp folder should actually have been removed, since we moved them to separate crates in the boards folder now. So: yes the crates from the boards directory, it will be easier to keep those up to date when updating to newer embassy versions.

abrudner, 2023-03-13:
Great, thanks for letting me know! Looking forward to using/contributing to this.

closed 2023-03-13 by abrudner