drogue-iot/drogue-device #374: Pico W example error linking

gilbertbw, 2023-08-02:
I am trying to get the example running on a Pico W, using the instuctions here: https://book.drogue.io/drogue-device/dev/examples/rp2040/pico-w/app/README.html

I am using the "Running with UF2 image" instuctions, and am getting an error on step 3

```
PS C:\GITRoot\drogue-device\examples\rp2040\pico-w\app> cargo build --release
   Compiling rp2040-pico-w v0.1.0 (C:\GITRoot\drogue-device\examples\rp2040\pico-w\app)
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "C:\\Users\\Gilbert\\AppData\\Local\\Temp\\rustcknm8GI\\symbols.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.0.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.1.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.10.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.11.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.12.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.13.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.14.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.15.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.2.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.3.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.4.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.5.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.6.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.7.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.8.rcgu.o" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_pico_w-4901ca84cba3eb70.rp2040_pico_w.e1835cce-cgu.9.rcgu.o" "--as-needed" "-L" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps" "-L" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\release\\deps" "-L" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\build\\rp2040-pico-w-78cc776101c1d519\\out" "-L" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\build\\cortex-m-7bcea78665f1282c\\out" "-L" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\build\\cortex-m-rt-a6d3bfd8a9140011\\out" "-L" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\build\\defmt-c50f544e84b25453\\out" "-L" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\build\\embassy-rp-da2508194ddd4bc7\\out" "-L" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\build\\rp2040-pac2-93954b094293253f\\out" "-L" "C:\\Users\\Gilbert\\.rustup\\toolchains\\nightly-2023-02-07-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_boot_rp-3b8b1b98056898f6.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libserde_json_core-8de27e85733a083a.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libryu-9273cabd98c5b5ec.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libcyw43-fd98b42ede813140.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libnum_enum-eb7f19c44c112c4c.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libcortex_m_rt-27c0a11f7ef31675.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_usb_logger-c223e127e73dd37f.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\liblog-515917fd7e895b99.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_usb-8cc0b6e8fb29b6d9.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libusbd_hid-7bddb39af41de508.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libusb_device-668fc9d89a4b396a.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libssmarshal-e24ee174b9cc5922.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libencode_unicode-cda6f2ee663585ca.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_net_driver_channel-7ce0e155af83d465.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libpanic_probe-2201b8d54e38b3b0.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_rp-3f07cbb5472bbfac.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembedded_hal_nb-7c8dc5d0c0eecf02.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\librp2040_pac2-5aeb5d87d9b66b0c.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_usb_driver-62af0156296510c7.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_futures-07462db31eea6cf9.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_cortex_m-7e833fa7853f48a3.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libcortex_m-4a1efc5894d96eda.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libvolatile_register-5397e4c54c66d382.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libvcell-cc695925722ec002.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libbare_metal-19182d66d9d57fca.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_net-8231b1e3837ae3d1.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libembassy_hal_common-ac156deafc98c0a2.rlib" "C:\\GITRoot\\drogue-device\\examples\\rp2040\\pico-w\\app\\target\\thumbv6m-none-eabi\\release\\deps\\libnum_traits-a0b286fd76d7eb0c.rlib" "C [... cut ...]
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 103003 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 103012 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 103040 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 103056 bytes
          rust-lld: error: too many errors emitted, stopping now (use --error-limit=0 to see all errors)

error: could not compile `rp2040-pico-w` due to previous error
```

lulf, 2023-08-02:
It looks like the wifi binary has increased in size. Could you try to make the following changes to the memory.x file?

```
-  FLASH                             : ORIGIN = 0x10007000, LENGTH = 512K
-  DFU                               : ORIGIN = 0x10087000, LENGTH = 516K
+  FLASH                             : ORIGIN = 0x10007000, LENGTH = 768K
+  DFU                               : ORIGIN = 0x100c7000, LENGTH = 772K
```

gilbertbw, 2023-08-03:
Thanks for that. Changing the `memory.x` file as suggested got it to build, but it's not running correctly.

I have figured out this line is never completing:
```rust
let (net_device, mut control, runner) = cyw43::new(state, pwr, spi, fw).await;
```

lulf, 2023-08-03:
The example is using a slightly outdated cyw43 driver. You might have a better chance if you try the wifi examples from embassy and adapt them to your need:

https://github.com/embassy-rs/embassy/tree/main/examples/rp
https://github.com/embassy-rs/embassy/blob/main/examples/rp/src/bin/wifi_blinky.rs