drogue-iot/drogue-device #376: Device Request:  Esp32-C3 Risc-V target

elasticdotventures, 2023-09-17:
👍🏻
I'd like to try and understand what will involved in getting Drogue-IOT integrated with the esp-rs ESP Rust Embedded community. 

I see this project already uses Embassy, so this presentation might be useful:
https://www.slideshare.net/georgiksk/rust-embedded-development-on-esp32-and-basics-of-async-with-embassy

elasticdotventures, 2023-09-17:
Also, tagging @georgik 😅

georgik, 2023-09-18:
@elasticdotventures Embassy and Async are supported. The only missing pieces are probably BSPs to get the project runnning:  https://github.com/drogue-iot/drogue-device/tree/main/boards
This can be probably derived from C version of BSPs for Espressif Devkits: https://github.com/espressif/esp-bsp
Or other source of inspiration might be DeviceScript - BSPs https://microsoft.github.io/devicescript/devices/esp32

lulf, 2023-09-19:
I think that's a great idea. I managed to explore this a while back but didn't get to the state of completion. I agree with @georgik that the missing piece is a BSP + an example.

elasticdotventures, 2023-09-29:
WIP https://github.com/drogue-iot/drogue-device/pull/388